nihalpasham/rustBoot #4: how to build Rustboot for nrf52840 Development kit.

datnguyentien1981, 2021-11-29:
Dear Nihal

 I have one nrf52840 kit  purchased at VietNam (https://infocenter.nordicsemi.com/pdf/nRF52840_DK_User_Guide_v1.2.pdf) 
 and I try to modify **test-firmware** and build **Rustboot** with below command

**cargo xtask build-sign-flash rustBoot nrf52840**

as your comment in
**https://github.com/nihalpasham/rustBoot/tree/main/boards/test_impls/nrf52840**
but currently, I have an issue as below.

> nad6hc@VM-NAD6HC-OSD5:~/Prj_Security/SourceCode/rustBoot-main$ **cargo xtask build-sign-flash rustBoot nrf52840**
>    Compiling version_check v0.9.3
>    Compiling typenum v1.14.0
>    Compiling p256 v0.9.0
>    Compiling defmt-macros v0.2.3
>    .....................
>    Compiling rustBoot v0.1.0 (/home/nad6hc/Prj_Security/SourceCode/rustBoot-main/rustBoot)
> 
> **error[E0554]: `#![feature]` may not be used on the stable release channel**
> --> rustBoot/src/lib.rs:3:1
>   |
> **3 | #![feature(once_cell)]**
>   | ^^^^^^^^^^^^^^^^^^^^^^
> For more information about this error, try `rustc --explain E0554`.
> 
> error: could not compile `rustBoot` due to previous error
> nad6hc@VM-NAD6HC-OSD5:~/Prj_Security/SourceCode/rustBoot-main$

 **could you please support me fix the issue in Rustboot.**

 

Dat Nguyen

nihalpasham, 2021-11-30:
> error[E0554]: #![feature] may not be used on the stable release channel

This is a rust compiler error (i.e. not a rustBoot error). You're probably using the stable channel for your builds. You need to switch to nightly. Here is a link to help you get rid of the error.

https://stackoverflow.com/questions/58226545/how-to-switch-between-rust-toolchains

datnguyentien1981, 2021-12-06:
Dear Nihal

I setup the environment as your advice with below configuration

> **rustup override set nightly**

and I can compile the Rustboot source code without error

> nad6hc@VM-NAD6HC-OSD5:~/Prj_Security/SourceCode/rustBoot-main/rustBoot$ cargo xtask build-sign-flash rustBoot nrf52840
>    Compiling version_check v0.9.3
>    Compiling typenum v1.14.0
>    Compiling subtle v2.4.1
>    Compiling rand_core v0.6.3
>    Compiling proc-macro2 v1.0.29
>    Compiling unicode-xid v0.2.2
>    Compiling semver v1.0.4
>    Compiling syn v1.0.76
>    Compiling zeroize v1.4.1
>    Compiling memchr v2.3.4
>    Compiling der v0.4.3
>    Compiling cpufeatures v0.2.1
>    Compiling defmt-parser v0.2.2
>    Compiling opaque-debug v0.3.0
>    Compiling cfg-if v1.0.0
>    Compiling anyhow v1.0.44
>    Compiling xshell-macros v0.1.17
>    Compiling ff v0.10.1
>    Compiling generic-array v0.14.4
>    Compiling nom v6.2.1
>    Compiling group v0.10.0
>    Compiling xshell v0.1.17
>    Compiling quote v1.0.9
>    Compiling defmt v0.2.3
>    Compiling digest v0.9.0
>    Compiling crypto-bigint v0.2.9
>    Compiling crypto-mac v0.11.1
>    Compiling block-buffer v0.9.0
>    Compiling signature v1.3.1
>    Compiling hmac v0.11.0
>    Compiling sha2 v0.9.8
>    Compiling elliptic-curve v0.10.6
>    Compiling ecdsa v0.12.4
>    Compiling p256 v0.9.0
>    Compiling defmt-macros v0.2.3
>    Compiling rustBoot v0.1.0 (/home/nad6hc/Prj_Security/SourceCode/rustBoot-main/rustBoot)
>    Compiling xtask v0.1.0 (/home/nad6hc/Prj_Security/SourceCode/rustBoot-main/xtask)
>     Finished dev [unoptimized + debuginfo] target(s) in 11.52s
>      Running `/home/nad6hc/Prj_Security/SourceCode/rustBoot-main/target/debug/xtask build-sign-flash rustBoot nrf52840`
> $ cargo build --release
>     Finished release [optimized + debuginfo] target(s) in 0.03s
> $ cargo build --release
>     Finished release [optimized + debuginfo] target(s) in 0.03s
> $ cargo build --release
>     Finished release [optimized + debuginfo] target(s) in 0.04s
> $ python3 convert2bin.py
> $ wsl python3 signer.py

Thank you
Dat Nguyen

nihalpasham, 2021-12-12:
Cool, I'm closing this issue.

closed 2021-12-12 by nihalpasham