nihalpasham/rustBoot #7: Unknown command: python3 in wsl

datnguyentien1981, 2021-12-20:
Dear Nihal

I have an issue related to wsl and python3. when i run command

> **wsl python3 signer.py**

and issue happens as below

> $ python3 convert2bin.py
> $ wsl python3 signer.py
> Unknown command: python3
> WSL
> Wsman Shell commandLine, version 0.2.1
> 
> USAGE: wsl COMMAND [PARAMS...]
> 
> COMMANDS:
> identify  - WS-Identify
> enum      - WS-Enumerate
> get       - WS-Get
> put       - WS-Put
> invoke    - WS-Invoke
> xclean    - Delete all files generated by this tool set
> xcred     - Create or display credential file
> xcert     - Get server certificate (saved to <IPADDRESS>.crt)
> 
> PARAMS specification is specific to a COMMAND.
> 
> Output will be saved to ./response.xml. If you want to run parallel
> executions in the same directory, define RTFILEPREFIX in the environment.
> Doing so may significantly increase files generated.
> 
> Requires: curl, xmllint, GNU core utilities.
> Optional: xsltproc for output formatting, gpg for encrypted credential.
> Optional: wget as alternate for curl when not available.
> Error: command `wsl python3 signer.py` failed, exit status: 1
> home@home:/media/home/3BBA6BB21611231A/data/Prj_Security/SourceCode/rustBoot-main/rustBoot$ python --version
> Python 3.6.9
> 

I think python3  is not recognized in wsl even python3 is installed

> https://www.codegrepper.com/code-examples/python/wsl+install+python+3.10
> https://www.codegrepper.com/code-examples/shell/bash%3A+python%3A+command+not+found

Could you please support me fix issue related to wsl and python3

Dat Nguyen

datnguyentien1981, 2022-01-03:
I try to compile rustboot on native ubuntu 18.4. So, I think there is no need to use wsl command (Windows Subsystem for Linux) and to avoid this  issue I remove wsl command.

in main.rs file (rustBoot-main/xtask/src/main.rs)

remove wsl command.
   //cmd!("wsl python3 signer.py").run()?;
     cmd!("python3 signer.py").run()?;

Dat Nguyen

nihalpasham, 2022-01-10:
> Dear Nihal
> 
> I have an issue related to wsl and python3. when i run command
> 
> > **wsl python3 signer.py**
> 
> and issue happens as below
> 
> > $ python3 convert2bin.py
> > $ wsl python3 signer.py
> > Unknown command: python3
> > WSL
> > Wsman Shell commandLine, version 0.2.1
> > USAGE: wsl COMMAND [PARAMS...]
> > COMMANDS:
> > identify  - WS-Identify
> > enum      - WS-Enumerate
> > get       - WS-Get
> > put       - WS-Put
> > invoke    - WS-Invoke
> > xclean    - Delete all files generated by this tool set
> > xcred     - Create or display credential file
> > xcert     - Get server certificate (saved to .crt)
> > PARAMS specification is specific to a COMMAND.
> > Output will be saved to ./response.xml. If you want to run parallel
> > executions in the same directory, define RTFILEPREFIX in the environment.
> > Doing so may significantly increase files generated.
> > Requires: curl, xmllint, GNU core utilities.
> > Optional: xsltproc for output formatting, gpg for encrypted credential.
> > Optional: wget as alternate for curl when not available.
> > Error: command `wsl python3 signer.py` failed, exit status: 1
> > home@home:/media/home/3BBA6BB21611231A/data/Prj_Security/SourceCode/rustBoot-main/rustBoot$ python --version
> > Python 3.6.9
> 
> I think python3 is not recognized in wsl even python3 is installed
> 
> > https://www.codegrepper.com/code-examples/python/wsl+install+python+3.10
> > https://www.codegrepper.com/code-examples/shell/bash%3A+python%3A+command+not+found
> 
> Could you please support me fix issue related to wsl and python3
> 
> Dat Nguyen

Just got back from my vacation. So, were you able to fix this? WSL is basically a light-weight VM that runs in Windows. We use WSL on windows as the python script has a few dependencies (i.e. imports `wolfcrypt` libraries) that need `Linux`.

nihalpasham, 2022-01-10:
> I try to compile rustboot on native ubuntu 18.4. So, I think there is no need to use wsl command (Windows Subsystem for Linux) and to avoid this issue I remove wsl command.
>
> in main.rs file [...]

Yeah, Ubuntu should work without issues. But you may need to edit the `main build` file in `xtask` to work in Ubuntu. The current implementation of xtask was only tested on Windows.

closed 2022-02-02 by nihalpasham