nihalpasham/rustBoot #9: imports wolfcrypt libraries

datnguyentien1981, 2022-01-10:
Dear Nihal

Currently, I have an issue related to wolfcrypt libraries when I compile Rustboot on native ubuntu.

```
home@home:~/Prj_Security/SourceCode/rustBoot-main/rustBoot$ rustup override set nightly
info: using existing install for 'nightly-x86_64-unknown-linux-gnu'
info: override toolchain for '/home/home/Prj_Security/SourceCode/rustBoot-main/rustBoot' set to 'nightly-x86_64-unknown-linux-gnu'
nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.59.0-nightly (f8abed9ed 2021-12-26)
home@home:~/Prj_Security/SourceCode/rustBoot-main/rustBoot$ cargo xtask build-sign-flash rustBoot nrf52840
Finished dev [unoptimized + debuginfo] target(s) in 1.07s
Running `/home/home/Prj_Security/SourceCode/rustBoot-main/target/debug/xtask build-sign-flash rustBoot nrf52840`
$ cargo build --release
    Finished release [optimized + debuginfo] target(s) in 0.74s
$ cargo build --release
    Finished release [optimized + debuginfo] target(s) in 0.04s
$ cargo build --release
    Finished release [optimized + debuginfo] target(s) in 0.08s
$ python3 convert2bin.py
$ python3 signer.py
Traceback (most recent call last):
File "sign.py", line 28, in <module>
from wolfcrypt import ciphers, hashes
ModuleNotFoundError: No module named 'wolfcrypt'
Traceback (most recent call last):
File "sign.py", line 28, in <module>
 from wolfcrypt import ciphers, hashes
ModuleNotFoundError: No module named 'wolfcrypt'
$ pyocd erase -t nrf52 --mass-erase
usage: pyocd [-h] [-V] [--help-options]  ...
pyocd: error: unrecognized arguments: --mass-erase
Error: command `pyocd erase -t nrf52 --mass-erase` failed, exit status: 2

```
As your recommendation in previous topic, I try to install wolfCrypt with below command
`pip install wolfcrypt`
but issue occurs as below

```
home@home:~/Prj_Security/SourceCode/rustBoot-main/rustBoot$ pip install wolfcrypt
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: wolfcrypt in /home/home/.local/lib/python3.6/site-packages (4.1.0.post0)
Requirement already satisfied: cffi in /home/home/.local/lib/python3.6/site-packages (from wolfcrypt) (1.15.0)
Requirement already satisfied: pycparser in /home/home/.local/lib/python3.6/site-packages (from cffi->wolfcrypt) (2.21)
```

I try to work around with another command
https://github.com/wolfSSL/wolfBoot/blob/master/docs/Signing.md#install-python3

```
git clone https://github.com/wolfSSL/wolfssl.git
cd wolfssl
./configure --enable-keygen --enable-rsa --enable-ecc --enable-ed25519 --enable-ed448 --enable-des3 CFLAGS="-DWOLFSSL_PUBLIC_MP"
sudo make install
```

and issue occurs as below

```
home@home:~/Prj_Security/SourceCode/rustBoot-main/rustBoot/wolfssl$ ./configure --enable-keygen --enable-rsa --enable-ecc --enable-ed25519 --enable-ed448 --enable-des3 CFLAGS="-DWOLFSSL_PUBLIC_MP"
bash: ./configure: No such file or directory
home@home:~/Prj_Security/SourceCode/rustBoot-main/rustBoot/wolfssl$ ls
async-check.sh   examples              m4                  scripts             wolfcrypt
AUTHORS          fips-check.sh         Makefile.am         SCRIPTS-LIST        wolfssl
autogen.sh       fips-hash.sh          mcapi               src                 wolfssl64.sln
certs            gencertbuf.pl         mplabx              sslSniffer          wolfssl.rc
ChangeLog.md     IDE                   mqx                 stamp-h.in          wolfssl.sln
cmake            input                 pre-commit.sh       support             wolfssl.vcproj
CMakeLists.txt   INSTALL               pre-push.sh         tests               wolfssl.vcxproj
commit-tests.sh  IPP                   pull_to_vagrant.sh  testsuite           wrapper
configure.ac     lib                   quit                tirtos              zephyr
COPYING          LICENSING             README              Vagrantfile
ctaocrypt        linuxkm               README.md           valgrind-bash.supp
cyassl           LPCExpresso.cproject  resource.h          valgrind-error.sh
doc              LPCExpresso.project   rpm                 wnr-example.conf
```

Could you please support me fix issue related to wolfcrypt libraries

Thank you

nihalpasham, 2022-02-02:
Can you try importing `wolfcrypt` from the python repl? I think you have a `path` issue of some sort.

datnguyentien1981, 2022-02-14:
Thank you for your support. after I fix path issue, I can install wolfcrypt with below command

> pip install wolfcrypt

> nad6hc@VM-NAD6HC-OSD5:~/Prj_Security/SourceCode/rustBoot-main/rustBoot$ pip install wolfcrypt
> 
> Collecting wolfcrypt
> 
>   Using cached https://rb-artifactory.bosch.com/artifactory/api/pypi/python-virtual/packages/packages/2d/5c/cb082c0e18670dc917aa58bc69bc8c12d9766775c526aa9547fca29f2d9e/wolfcrypt-4.1.0.post0-cp27-cp27mu-manylinux1_x86_64.whl
> 
> Collecting cffi (from wolfcrypt)
> 
>   Using cached https://rb-artifactory.bosch.com/artifactory/api/pypi/python-virtual/packages/packages/26/28/fb01ff898aa7c93e4774799b3dc0a3693cfee48c5ea4e524ce30e6b10e7e/cffi-1.15.0-cp27-cp27mu-manylinux1_x86_64.whl
> 
> Collecting pycparser (from cffi->wolfcrypt)
> 
>   Using cached https://rb-artifactory.bosch.com/artifactory/api/pypi/python-virtual/packages/packages/62/d5/5f610ebe421e85889f2e55e33b7f9a6795bd982198517d912eb1c76e1a53/pycparser-2.21-py2.py3-none-any.whl
> 
> Installing collected packages: pycparser, cffi, wolfcrypt
> 
> Successfully installed cffi-1.15.0 pycparser-2.21 wolfcrypt-4.1.0.post0.

Dat Nguyen

closed 2022-02-14 by datnguyentien1981