nihalpasham/rustBoot #36: Order IMX-8 chip set.

Strange21, 2022-06-21:
Need to order the IMX8 series of processor for our TOP97 client. 

- [ ] Check online for the vendor with whom the required chip is available.
- [ ] Check with our Internal logistics team to get the estimated time of delivery
- [ ] Place the order.

nihalpasham, 2022-06-24:
Can you add a link to the list of boards (or options)?

Strange21, 2022-06-24:
Following is the list of the chips
https://www.digikey.in/en/product-highlight/n/nxp-semi/imx-8-series-applications-processors

nihalpasham, 2022-06-28:
So, we'll need a `NXP I.MX8M-nano` board - https://www.digikey.in/en/product-highlight/n/nxp-semi/imx-8-series-applications-processors#imx8nano

Here's an example of an i.mx8m-nano board
- `variscite i.mx8m-starter kit` - https://www.variscite.com/product/evaluation-kits/var-som-mx8m-nano-evaluation-kits/

Strange21, 2022-07-05:
I came across following 2 evaluation kit. we can select any one from it. 
[nxp-8mnanod4evk-eval-kit](https://www.mouser.in/new/nxp-semiconductors/nxp-8mnanod4evk-eval-kit/)
[nxp-imx-8m-nano-ultralite-eval-kit](https://www.mouser.in/new/nxp-semiconductors/nxp-imx-8m-nano-ultralite-eval-kit/)

As Mouser is already our registered vendor therefore we can prefer them.
[nxp-8mnanod4evk-eval-kit](https://www.mouser.in/new/nxp-semiconductors/nxp-8mnanod4evk-eval-kit/) is readily available for dispatch which might take 2-3 weeks to get delivered.

nihalpasham, 2022-07-13:
lets put in an order for the first one - [nxp-8mnanod4evk-eval-kit](https://www.mouser.in/new/nxp-semiconductors/nxp-8mnanod4evk-eval-kit/)

nihalpasham, 2022-07-22:
@Strange21 - any luck placing the order

Strange21, 2022-07-22:
So Rajkumar have Raised the Purchase request, he sent it for the Approval. 
It is costing about 40000/-.