nihalpasham/rustBoot #64: Add support for STM32F469 MCU in examples

lionelains, 2022-09-05:
Hello, and first of all, thanks for this great work.

I have tried to add support for the STM32F469 MCU in the STM HAL.
This chip is used on the [STM32F469-IDISCO board](https://www.st.com/en/evaluation-tools/32f469idiscovery.html) and supporting should be very close to the existing code implemented for STM32F411 and STM32F446.
However, I am wondering how to adjust the value for the constant `FW_RESET_VTR`, for which the definition is based on an offset...
[This offset is 0xA9 in the current STM446 implementation]
(https://github.com/nihalpasham/rustBoot/blob/5a3cc7aa392abeb0ac322aa4279bba5a8eba7c6e/boards/hal/src/stm/stm32f446.rs#L15)
[...and 0x99 in the STM411 implementation](https://github.com/nihalpasham/rustBoot/blob/5a3cc7aa392abeb0ac322aa4279bba5a8eba7c6e/boards/hal/src/stm/stm32f411.rs#L15).

What is the meaning of this offset? How can I adjust this for STM32F469?
In the [STM32 programming manual (PM201)](https://www.st.com/resource/en/programming_manual/pm0214-stm32-cortexm4-mcus-and-mpus-programming-manual-stmicroelectronics.pdf), the vector table is identical regardless of the chip STM32F411 or STM32F446...

lionelains, 2022-09-05:
Digging around the binaries produced by the STM32F411 or STM32F446 counterparts, it looks like this offset can be guessed by grabbing the 32-bit reset vector located at offset 4 in the binary image.
When I build for STM32F446, I indeed get a pointer to address 0x080202a9 (`BASE_ADDR` being 0x08020000), but `RB_HDR_SIZE` is not meaningful here because we haven't prepended the rustBoot header yet...

imrank03, 2022-09-05:
> Hello, and first of all, thanks for this great work.
>
> I have tried to add support for the STM32F469 MCU in the STM HAL. This chip is used on the [STM32F469-IDISCO [...]

Hi @lionelains,

Thanks for reaching out, and glad to hear that you're interested in rustBoot!

- `FW_RESET_VTR` is reset handler’s address located in the binary image (example: `BASE_ADDR` 0x08020000 + 4)
https://github.com/nihalpasham/rustBoot/blob/5a3cc7aa392abeb0ac322aa4279bba5a8eba7c6e/boards/hal/src/stm/stm32f446.rs#L15

imrank03, 2022-09-05:
> Digging around the binaries produced by the STM32F411 or STM32F446 counterparts, it looks like this offset can be guessed by grabbing the 32-bit reset vector located at offset 4 in the binary image. [...]

Yes, we have not prepended the rustBoot header yet in `HAL implementation` but  we are using `RB_HDR_SIZE` in the `boot_from` function.

https://github.com/nihalpasham/rustBoot/blob/5a3cc7aa392abeb0ac322aa4279bba5a8eba7c6e/boards/hal/src/stm/stm32f446.rs#L229

imrank03, 2022-09-05:
> Digging around the binaries produced by the STM32F411 or STM32F446 counterparts, it looks like this offset can be guessed by grabbing the 32-bit reset vector located at offset 4 in the binary image. [...]

Yes, we haven't prepended the rustBoot header yet, but we use `RB_HDR_SIZE` to create `FW_RESET_VTR` address used in `boot_from` function.

https://github.com/nihalpasham/rustBoot/blob/5a3cc7aa392abeb0ac322aa4279bba5a8eba7c6e/boards/hal/src/stm/stm32f446.rs#L229

lionelains, 2022-09-06:
Hi @imrank03.
Thanks for your reply.
OK, if I understand correctly your point, when rustBoot is running from flash, the partition where it should load the application firmware has the header prepended, so we indeed need to read `RB_HDR_SIZE` bytes away.

But, in that case, based on the fact that the vector table is directly located at the very beginning inside the application's binary image we should find :
* the initial SP value vector at offset +0 as 32-bit value
* the reset vector at offset +4 as 32-bit value

Thus, why don't we just have the following?
```rust
pub const FW_RESET_VTR    : u32 = BASE_ADDR + RB_HDR_SIZE + 4; 
```

imrank03, 2022-09-07:
We are checking the reset handler value at offset +4, `FW_RESET_VTR` must match the reset handler address, otherwise the **boot_from** function will panic.
https://github.com/nihalpasham/rustBoot/blob/5a3cc7aa392abeb0ac322aa4279bba5a8eba7c6e/boards/hal/src/stm/stm32f446.rs#L229-L230

> if we use this 
> ```rust
> pub const FW_RESET_VTR    : u32 = BASE_ADDR + RB_HDR_SIZE + 4; 
> ```
> will get wrong 32 bit value and it will be different from +4 offset value.

Here's a tiny [post](https://link.medium.com/zmv4Tf5hHbb) on why we're doing this and what `RefinedUsize` is.

lionelains, 2022-09-08:
Oh, I see, the `FW_RESET_VTR` is only a failsafe check, the real value to jump to is actually the 32-bit value provided to `boot_from()` inside its parameter `fw_base_address`.

lionelains, 2022-09-08:
If you're interested, I think I have a port of the rustBoot demo (boot firmware->upgrade firmware with LED associated blinks) working on the STM32F469-IDISCO board...
I can propose this as a PR.

_Please note that I didn't however implement any support for the external QSPI flash that is also mounted on that board, so I'm only supporting the built-in flash_

imrank03, 2022-09-10:
Hi @lionelains,

Thank you, we are completely open-source and as such, we welcome contributions from anyone! We would appreciate it if you proposed this as a PR.

lionelains, 2022-09-12:
Hi @imrank03, I have created PR https://github.com/nihalpasham/rustBoot/pull/65

imrank03, 2022-09-13:
Acknowledged

commit 146ebbfa (2022-09-15): Merge pull request #65 from lionelains/main

Add support for STM32F469 MCU in examples (closes #64)
closed 2022-09-15 by nihalpasham