nihalpasham/rustBoot #68: Add a rust-based  secure element "NXP SE050" driver for ARM TrustZone-M and A

imrank03, 2022-09-21:
Need to implement a secure element `NXP SE050` driver for **ARM TrustZone-M and A**.

**Sub-tasks:**

- [ ] I2C driver for NXP's crypto authentication HW (i.e. secure element `NXP SE050`).
- [ ] A method for sending commands to the `NXP SE050` and retrieving the associated response.
- [ ] **rustBoot** demonstration with secure element.

imrank03, 2024-11-26:
> Is there any code available for this issue? I would be interested in working on this

Hi @wackazong ,  I have worked on this driver, and here is the [link](https://github.com/imrank03/nxp-se050). I tested it on an STM32H723 board, but the driver is not fully written because I got busy with other projects. I believe that in the meantime, someone else has written the full driver. Here is the [link](https://github.com/sosthene-nitrokey/se050-generation/tree/main) for that. The thing is, this driver comes with a license that is not MIT or Apache. I plan to work on this issue and provide the driver under the MIT license. It would be great if you could contribute to this project and create a PR.

wackazong, 2024-11-26:
Yes, I found that now, too. Thanks. I believe the most current version is here: https://crates.io/crates/se05x